MiSTer-devel/SNES_MiSTer #427: Rockman X2 title screen has lag frames

AkiteruSDA, 2025-08-31:
Tested using core from 25.06.05. The title screen for Rockman X2 on MiSTer has lag frames when X shoots his buster shot, but doesn't on a real SNES and original cartridge. Here's a video comparison: https://www.youtube.com/watch?v=KgS3mVSKoK4

I know this is minor, but since there isn't much happening on the title screen, the presence of lag might point to some Cx4-related inaccuracy.

TheJesusFish, 2025-09-21:
Please check the latest unstable if you haven't already. https://github.com/MiSTer-unstable-nightlies/SNES_MiSTer/releases/download/unstable-builds/SNES_unstable_20250920_09815f.rbf

real-amano, 2025-09-28:
@AkiteruSDA did you check? If the unstable worked for you, please close the bug.

AkiteruSDA, 2025-10-01:
Yes, the issue still appears to be present in 20250928.